adilinden-oss/octoprint-webcamstreamer #1: Error when trying to stream

drkline, 2019-02-08:
Hello,

I thought that I had everything configured correctly but when I click on Go Live I get the error below.  Can you shed some insight as to what this error means?

Thank you,
Dave 

WebcamStreamer Error
'NoneType' object has no attribute 'containers'

adilinden, 2019-02-08:
I see that my instructions are incorrect!

Assuming you properly installed docker per below:

```
curl -sSL https://get.docker.com | sh
sudo usermod pi -aG docker
sudo reboot
```

The correct docker pull command should be:

```
docker pull adilinden/rpi-ffmpeg:latest
```

Adi

adilinden, 2019-02-08:
I just updated `README.md` with the correction.

darkstudio, 2019-02-09:
> 
> 
> Hello,
> 
> I thought that I had everything configured correctly but when I click on Go Live I get the error below. Can you shed some insight as to what this error means?
> 
> Thank you,
> Dave
> 
> WebcamStreamer Error
> 'NoneType' object has no attribute 'containers'

IDEM

darkstudio, 2019-02-09:
No error message anymore when we applied your command but doesnt work.  No stream on youtube....

adilinden, 2019-02-09:
Does IDEM mean "the same"? If so, can you run the following commands on the command line and tell me what the results are:

```docker image ls```

and

```docker container ls```

You should see something like this:

```
pi@octopi:~ $ docker image ls
REPOSITORY             TAG                 IMAGE ID            CREATED             SIZE
adilinden/rpi-ffmpeg   latest              3f83e20f5507        3 weeks ago         360MB
pi@octopi:~ $ docker container ls
CONTAINER ID        IMAGE                         COMMAND                  CREATED             STATUS              PORTS               NAMES
f8c932afaad5        adilinden/rpi-ffmpeg:latest   "/usr/bin/entry.sh f…"   2 hours ago         Up 2 hours                              WebStreamer
```

hapklaar, 2019-02-13:
I used your instructions to install docker, but prepended sudo for the curl cmd as well. All went without errors, but I don't seem to have a container:

`admin@HapPi3:~ $ sudo docker image ls
REPOSITORY             TAG                 IMAGE ID            CREATED             SIZE
adilinden/rpi-stream   latest              0d9e199dc4e2        5 hours ago         360MB
admin@HapPi3:~ $ sudo docker container ls
CONTAINER ID        IMAGE               COMMAND             CREATED             STATUS              PORTS               NAMES
admin@HapPi3:~ $`

hapklaar, 2019-02-13:
My issue was caused by the fact that I pulled the image as root. Fixed it by removing the docker installation and installed it again using:

sudo apt-get install docker-ce
sudo usermod <user that runs octoprint> -aG docker
docker pull adilinden/rpi-stream:latest

after this pulin would start correctly

adilinden, 2019-02-15:
Glad to hear it working for you!

closed 2019-02-15 by adilinden

DrPhreak85, 2021-02-21:
Hey Together,
after using 

`docker pull adilinden/rpi-ffmpeg:latest`

i get back the following:

`pi@octopi:~/docker-rpi-ffmpeg $ docker pull adilinden/rpi-ffmpeg:latest
Error response from daemon: pull access denied for adilinden/rpi-ffmpeg, repository does not exist or may require 'docker login': denied: requested access to the resource is denied`

After some research, i can't find the container on Docker.

Greets and 73 DrPhreak

DrPhreak85, 2021-02-21:
Hey me again,
also the install via ssh und build it from git fails:
`pi@octopi:~/docker-rpi-ffmpeg $ sudo docker build -t adilinden/rpi-ffmpeg .
Sending build context to Docker daemon  78.85kB
Step 1/16 : FROM balenalib/raspberry-pi-debian:stretch as builder
 ---> 1b831c517089
Step 2/16 : RUN apt-get update -qy && apt-get -qy install         build-essential git nasm         libomxil-bellagio-dev
 ---> Using cache
 ---> d27732dafdf8
Step 3/16 : WORKDIR /root
 ---> Using cache
 ---> 7ce184df545a
Step 4/16 : RUN git clone https://github.com/FFmpeg/FFmpeg.git --depth 1
 ---> Using cache
 ---> 3e3df3c1775c
Step 5/16 : WORKDIR /root/FFmpeg
 ---> Using cache
 ---> a76b723aa9f5
Step 6/16 : RUN ./configure --arch=armel --target-os=linux --enable-gpl --enable-omx --enable-omx-rpi --enable-nonfree
 ---> [Warning] The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm/v7) and no specific platform was requested
 ---> Running in f8052c9de396
ERROR: OpenMAX IL headers from raspberrypi/firmware not found

If you think configure made a mistake, make sure you are using the latest
version from Git.  If the latest version fails, report the problem to the
ffmpeg-user@ffmpeg.org mailing list or IRC #ffmpeg on irc.freenode.net.
Include the log file "ffbuild/config.log" produced by configure as this will help
solve the problem.
The command '/bin/sh -c ./configure --arch=armel --target-os=linux --enable-gpl --enable-omx --enable-omx-rpi --enable-nonfree' returned a non-zero code: 1
`

Is there an easy solution?

73 DrPhreak