adilinden-oss/octoprint-webcamstreamer #2: Live but not Live

Xlaits, 2019-02-08:
After setting up my everything, when I click go live, it goes live. But YouTube is not starting the stream at all. It's as if it's working on my end, but they're not receiving the packets on their end. Does my Pi need to be Port Forwarded or something for this to work properly?

adilinden, 2019-02-08:
Are you looking at your live stream in the Dashboard? If you are looking at the plugin tab, the stream has to be set to "Public" else it will report an error. I just experienced that yesterday with it set to "Unlisted" on my stream.

Xlaits, 2019-02-08:
On the YouTube Dash? Yes.

adilinden, 2019-02-09:
Can you run the following commands on the command line and tell me what the results are:

```docker image ls```

and

```docker container ls```

You should see something like this:

```
pi@octopi:~ $ docker image ls
REPOSITORY             TAG                 IMAGE ID            CREATED             SIZE
adilinden/rpi-ffmpeg   latest              3f83e20f5507        3 weeks ago         360MB
pi@octopi:~ $ docker container ls
CONTAINER ID        IMAGE                         COMMAND                  CREATED             STATUS              PORTS               NAMES
f8c932afaad5        adilinden/rpi-ffmpeg:latest   "/usr/bin/entry.sh f…"   2 hours ago         Up 2 hours                              WebStreamer
```

Xlaits, 2019-02-11:
`pi@octopi:~ $ docker image ls
REPOSITORY              TAG                 IMAGE ID            CREATED             SIZE
octoprint/youtubelive   latest              4a0d64874445        2 days ago          1.32GB
adilinden/rpi-stream    latest              f065a3ad0335        3 weeks ago         360MB
adilinden/rpi-ffmpeg    latest              3f83e20f5507        3 weeks ago         360MB
alexellis2/streaming    07-05-2018          a8d02138647f        9 months ago        1.32GB
`
`pi@octopi: ~ $ docker container ls
CONTAINER ID        IMAGE               COMMAND             CREATED             STATUS              PORTS               NAMES
`
It seems that there are no containers.

adilinden, 2019-02-13:
No container means that `ffmpeg` inside the container was unable to start. This is most likely because either the `Stream Server URL` or the `OctoPi Webcam URL` is incorrect.

For YouTube the `Stream Server URL` should be something like `rtmp://a.rtmp.youtube.com/live2/xxxx-xxxx-xxxx-xxxx`, where `xxxx-xxxx-xxxx-xxxx` needs to be replaced with your streaming key.

The `OctoPi Webcam URL` needs to be fully qualified. It cannot be `localhost` either. You would need to know the IP address for your OctoPi and construct the URL like this `http://192.168.0.10:8080/?action=stream` but where `192.168.0.10` is the actual IP address of the OctoPi.

Lastly, you should be able to startup the `adilinden/rpi-ffmpeg` container using a command line. It a long one, but it will work if all parameters are correct. An example would be:

```
docker run --privileged --name test  -ti -rm adilinden/rpi-ffmpeg:latest ffmpeg -re -f mjpeg -framerate 5 -i http://192.168.0.10:8080/?action=stream -ar 44100 -ac 2 -acodec pcm_s16le -f s16le -ac 2 -i /dev/zero -acodec aac -ab 128k -vcodec h264 -pix_fmt yuv420p -framerate 5 -g 10 -strict experimental -f flv rtmp://a.rtmp.youtube.com/live2/xxxx-xxxx-xxxx-xxxx
```
Again, you would need to replace `192.168.0.10` with your OctoPi IP and `xxxx-xxxx-xxxx-xxxx` with your YouTube Live Key.

I am working on updating the `rpi-stream` container to support this plugin and also introduce some options that would make testing much easier.

ksztibor, 2019-02-13:
Hi All,

@Xlaits 
I have had the same problem, and my case might help you. In my case the cause of the problem was the inappropriate definiton of the parameter 'OctoPi Webcam URL': I used 'https' instead of 'http', and I used the hostname instead of the IP address of the Octopi. With 'http' and with the IP address now it works for me. During the troubleshooting I also repeated the Installation process, but I'm not sure, whether it had any effect. 

@adilinden 
In the latest comment, in the example command line there is a typo: instead of '-rm' there should be '--rm' written:
> Lastly, you should be able to startup the `adilinden/rpi-ffmpeg` container using a command line. It a long one, but it will work if all parameters are correct. An example would be:
>
> ```
> docker run [...]

Xlaits, 2019-02-14:
Ran the above command, with the fix that @ksztibor mentioned. Below is the output of that command.
```
pi@octopi:~ $ docker run --privileged --name test  -ti --rm adilinden/rpi-ffmpeg:latest ffmpeg -re -f mjpeg -framerate 30 -i http://192.168.1.14:8080/?action=stream -ar 44100 -ac 2 -acodec pcm_s16le -f s16le -ac 2 -i /dev/zero -acodec aac -ab 128k -vcodec h264 -pix_fmt yuv420p -framerate 30 -g 10 -strict experimental -f flv rtmp://a.rtmp.youtube.com/live2/xxxx-xxxx-xxxx-xxxx
ffmpeg version git-2019-01-19-32fb83e Copyright (c) 2000-2019 the FFmpeg developers
  built with gcc 6.3.0 (Raspbian 6.3.0-18+rpi1+deb9u1) 20170516
  configuration: --arch=armel --target-os=linux --enable-gpl --enable-omx --enable-omx-rpi --enable-nonfree
  libavutil      56. 26.100 / 56. 26.100
  libavcodec     58. 44.100 / 58. 44.100
  libavformat    58. 26.100 / 58. 26.100
  libavdevice    58.  6.101 / 58.  6.101
  libavfilter     7. 48.100 /  7. 48.100
  libswscale      5.  4.100 /  5.  4.100
  libswresample   3.  4.100 /  3.  4.100
  libpostproc    55.  4.100 / 55.  4.100
[mjpeg @ 0x2355f70] unable to decode APP fields: Invalid data found when processing input
Input #0, mjpeg, from 'http://192.168.1.14:8080/?action=stream':
  Duration: N/A, bitrate: N/A
    Stream #0:0: Video: mjpeg (Baseline), yuvj422p(pc, bt470bg/unknown/unknown), 640x480, 30 tbr, 1200k tbn, 30 tbc
Guessed Channel Layout for Input Stream #1.0 : stereo
Input #1, s16le, from '/dev/zero':
  Duration: N/A, bitrate: 1411 kb/s
    Stream #1:0: Audio: pcm_s16le, 44100 Hz, stereo, s16, 1411 kb/s
Stream mapping:
  Stream #0:0 -> #0:0 (mjpeg (native) -> h264 (h264_omx))
  Stream #1:0 -> #0:1 (pcm_s16le (native) -> aac (native))
Press [q] to stop, [?] for help
[mjpeg @ 0x2357f00] unable to decode APP fields: Invalid data found when processing input
[swscaler @ 0x23c7fd0] deprecated pixel format used, make sure you did set range correctly
[h264_omx @ 0x23acdc0] Using OMX.broadcom.video_encode
Output #0, flv, to 'rtmp://a.rtmp.youtube.com/live2/1wa2-hc72-kqu6-0jwq':
  Metadata:
    encoder         : Lavf58.26.100
    Stream #0:0: Video: h264 (h264_omx) ([7][0][0][0] / 0x0007), yuv420p(progressive), 640x480, q=2-31, 200 kb/s, 30 fps, 1k tbn, 30 tbc
    Metadata:
      encoder         : Lavc58.44.100 h264_omx
    Stream #0:1: Audio: aac (LC) ([10][0][0][0] / 0x000A), 44100 Hz, stereo, fltp, 128 kb/s
    Metadata:
      encoder         : Lavc58.44.100 aac
[mjpeg @ 0x2357f00] unable to decode APP fields: Invalid data found when processing input
    Last message repeated 1 times
frame=    3 fps=0.0 q=-0.0 size=       1kB time=00:00:00.04 bitrate= 255.0kbits/[mjpeg @ 0x2357f00] unable to decode APP fields: Invalid data found when processing input
    Last message repeated 11 times
frame=   15 fps= 15 q=-0.0 size=      12kB time=00:00:00.46 bitrate= 202.9kbits/[mjpeg @ 0x2357f00] unable to decode APP fields: Invalid data found when processing input
    Last message repeated 4 times
frame=   20 fps= 13 q=-0.0 size=      16kB time=00:00:00.62 bitrate= 207.4kbits/[mjpeg @ 0x2357f00] unable to decode APP fields: Invalid data found when processing input
    Last message repeated 4 times
frame=   25 fps= 12 q=-0.0 size=      21kB time=00:00:00.81 bitrate= 214.8kbits/[mjpeg @ 0x2357f00] unable to decode APP fields: Invalid data found when processing input
    Last message repeated 4 times
frame=   30 fps= 12 q=-0.0 size=      25kB time=00:00:00.97 bitrate= 207.5kbits/[mjpeg @ 0x2357f00] unable to decode APP fields: Invalid data found when processing input
    Last message repeated 4 times
frame=   35 fps= 11 q=-0.0 size=      31kB time=00:00:01.13 bitrate= 223.3kbits/[mjpeg @ 0x2357f00] unable to decode APP fields: Invalid data found when processing input
    Last message repeated 4 times
frame=   40 fps= 11 q=-0.0 size=      34kB time=00:00:01.30 bitrate= 214.9kbits/[mjpeg @ 0x2357f00] unable to decode APP fields: Invalid data found when processing input
    Last message repeated 4 times
frame=   45 fps= 11 q=-0.0 size=      40kB time=00:00:01.46 bitrate= 222.4kbits/[mjpeg @ 0x2357f00] unable to decode APP fields: Invalid data found when processing input
    Last message repeated 4 times
frame=   50 fps= 11 q=-0.0 size=      43kB time=00:00:01.62 bitrate= 215.7kbits/[mjpeg @ 0x2357f00] unable to decode APP fields: Invalid data found when processing input
    Last message repeated 4 times
frame=   55 fps= 11 q=-0.0 size=      49kB time=00:00:01.81 bitrate= 219.7kbits/[mjpeg @ 0x2357f00] unable to decode APP fields: Invalid data found when processing input
    Last message repeated 4 times
frame=   60 fps= 11 q=-0.0 size=      52kB time=00:00:01.97 bitrate= 214.2kbits/[mjpeg @ 0x2357f00] unable to decode APP fields: Invalid data found when processing input
    Last message repeated 4 times
frame=   65 fps= 11 q=-0.0 size=      56kB time=00:00:02.13 bitrate= 213.1kbits/[mjpeg @ 0x2357f00] unable to decode APP fields: Invalid data found when processing input
    Last message repeated 4 times
frame=   70 fps= 11 q=-0.0 size=      61kB time=00:00:02.29 bitrate= 216.3kbits/[mjpeg @ 0x2357f00] unable to decode APP fields: Invalid data found when processing input
    Last message repeated 4 times
frame=   75 fps= 11 q=-0.0 size=      67kB time=00:00:02.46 bitrate= 222.1kbits/[mjpeg @ 0x2357f00] unable to decode APP fields: Invalid data found when processing input
    Last message repeated 4 times
[... cut ...]
frame=  497 fps= 10 q=-0.0 size=     430kB time=00:00:16.53 bitrate= 213.1kbits/[mjpeg @ 0x2357f00] unable to decode APP fields: Invalid data found when processing input
    Last message repeated 2 times
[flv @ 0x23abf10] Failed to update header with correct duration.
[flv @ 0x23abf10] Failed to update header with correct filesize.
frame=  500 fps= 10 q=-0.0 Lsize=     433kB time=00:00:16.67 bitrate= 212.7kbits/s speed=0.334x
video:407kB audio:4kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 5.380288%
[aac @ 0x23ae300] Qavg: 65536.000
Exiting normally, received signal 2.

```

churusaa, 2019-02-14:
I'm having a similar issue with my instance, but I have HSTS set for my domain, so I actually do require https:// support, which is apparently not baked into the current build being published. Would you be able to compile with HTTPS?

adilinden, 2019-02-15:
@Xlaits When you ran `docker run --privileged --name test  -ti --rm adilinden/rpi-ffmpeg:latest ffmpeg -re -f mjpeg -framerate 30 -i http://192.168.1.14:8080/?action=stream -ar 44100 -ac 2 -acodec pcm_s16le -f s16le -ac 2 -i /dev/zero -acodec aac -ab 128k -vcodec h264 -pix_fmt yuv420p -framerate 30 -g 10 -strict experimental -f flv rtmp://a.rtmp.youtube.com/live2/xxxx-xxxx-xxxx-xxxx` were you able to see the stream in the YouTube Live Dashboard? Also, did you replace `xxxx-xxxx-xxxx-xxxx` with your actual key?

Xlaits, 2019-02-15:
Yes, and yes. However, running that command just spits out tons of errors
to the cli.

adilinden, 2019-02-16:
Hi @Xlaits 

I see your command line says:

```
docker run --privileged --name test  -ti --rm adilinden/rpi-ffmpeg:latest ffmpeg -re -f mjpeg -framerate 30 -i http://192.168.1.14:8080/?action=stream -ar 44100 -ac 2 -acodec pcm_s16le -f s16le -ac 2 -i /dev/zero -acodec aac -ab 128k -vcodec h264 -pix_fmt yuv420p -framerate 30 -g 10 -strict experimental -f flv rtmp://a.rtmp.youtube.com/live2/xxxx-xxxx-xxxx-xxxx
```

You also have in there `-framerate 30` and `-framerate 30 -g 10`. I have not been able to make rPi work smoothly with a framerate higher than the default `5` or resolution higher than the default. At least on the rPi 2B that I am using the CPU is insufficient. Also, the ffmpeg guide states that `-gop` should be double the framerate. This means you should have `-framerate 30 -g 60` per [FFmpeg Streaming Guide](https://trac.ffmpeg.org/wiki/StreamingGuide).

Adi

Xlaits, 2019-02-16:
I'm using one of the latest and greatest RPi boards, and a USB webcam. I'm still getting tons of errors popping up, and my cli is locked down once that command is run. That invalid data thing keeps running and running...

adilinden, 2019-02-18:
Hi @Xlaits 

I understand from others that have tried, that even the rPi 3B is unable to cope with anything more than 640x480 and 5fps. Sorry...

Adi

Xlaits, 2019-02-18:
Either way, I have got it working just fine, but it doesn't auto-start the stream when I start printing.

Xlaits, 2019-02-20:
So, I had https in the settings on the octoprint setup. Do'h.

closed 2019-02-20 by Xlaits