adilinden-oss/octoprint-webcamstreamer #3: issue with buffering on twitch

lowestlime76, 2019-02-10:
So I was able to get webcamstreamer to go live, but when it does all it will do is buffer? Anything I could have missed. I running a raspberry 3b+. Any suggestions would be great.

adilinden, 2019-02-11:
Are you looking at the webcamstreamer tab in OctoPrint or the stream on the Twitch website?

lowestlime76, 2019-02-11:
Both.

adilinden, 2019-02-13:
While streaming, can you run the commands described in post https://github.com/adilinden-oss/octoprint-webcamstreamer/issues/2#issuecomment-462087802. The commands are:

```
docker image ls
docker container ls
```

The first command should include the `adilinden/rpi-ffmpeg` image and the second command should include a running `adilinden/rpi-ffmpeg` image.

lowestlime76, 2019-02-18:
pi@octopiender1:~ $ docker image ls
REPOSITORY             TAG                 IMAGE ID            CREATED                                                                                                                          SIZE
adilinden/rpi-stream   latest              f065a3ad0335        4 weeks ago                                                                                                                      360MB
adilinden/rpi-ffmpeg   latest              3f83e20f5507        4 weeks ago                                                                                                                      360MB
pi@octopiender1:~ $ docker container ls
CONTAINER ID        IMAGE                         COMMAND                  CREAT                                                                                                             ED             STATUS              PORTS               NAMES
ae77f6ad876d        adilinden/rpi-ffmpeg:latest   "/usr/bin/entry.sh f…"   16 mi                                                                                                             nutes ago      Up 16 minutes                           WebStreamer

adilinden, 2019-02-18:
Is your rPi cam 640x480 with 5fps? The default plugin settings assume those settings for the rPi cam.

lowestlime76, 2019-02-18:
I'm using a USB webcam. Logitech c270.