adilinden-oss/octoprint-webcamstreamer #4: Unable to start

jneilliii, 2019-02-13:
I have yet to do any searching on this specific error, but wanted to post here as a placeholder.  Followed install instructions exactly as described, configured the settings and upon start get this error.  
```
500 Server Error: Internal Server Error ("failed to create endpoint WebStreamer on network bridge: failed to add the host (veth099bf05) <=> sandbox (vethcba8c9c) pair interfaces: operation not supported")
```
Receive same error when trying to run the command directly
```
docker run --privileged --name test  -ti adilinden/rpi-ffmpeg:latest ffmpeg -re -f mjpeg -framerate 5 -i http://192.168.0.120:8080/?action=stream -ar 44100 -ac 2 -acodec pcm_s16le -f s16le -ac 2 -i /dev/zero -acodec aac -ab 128k -vcodec h264 -pix_fmt yuv420p -framerate 5 -g 10 -strict experimental -f flv rtmp://a.rtmp.youtube.com/live2/xxxx-xxxx-xxxx-xxxx

```

adilinden, 2019-02-15:
@jneilliii I believe this is a docker issue related to docker networking. I did a quick Google search but it is not bringing up anything helpful. Unfortunately I don't know enough about docker to provide any meaningful troubleshooting advise there.

jneilliii, 2019-02-15:
So, I've been able to run the test command if I include `--network=host` in the docker command line.  I believe that will also allow the use of `localhost` and `127.0.0.1` for the input streams based on a PR that was submitted to one of my plugins.  I think there would be a small change necessary on the plugin side to accommodate this. May submit a PR to your repo for the change if that's ok?

PR #6 (2019-02-16): Added `network_mode = "host"` to the docker container run command.
This worked for me to resolve #4. I don't think there would be any adverse effects with this for anyone else and might have the added benefit of using 127.0.0.1 or localhost for stream sources.
closed 2019-02-16 by adilinden