adilinden-oss/octoprint-webcamstreamer #5: choppy video with high latency

ksztibor, 2019-02-13:
Hi Adi,

using the default ffmpeg command (and the predefined framerate 5), in Youtube I have very high latency (about 30 seconds) and choppy video (it seems to be dropping a lot of frames). How high is the latency in your case? Is the video continuous for you?  I'm not sure, whether the latency is because of ffmpeg on the client side, or it is on the youtube side. Do you know, how the ffmpeg stream can be checked locally? I've tried VLC, but I couldn't manage it.

By the way, my mjpeg-streamer is set to resolution 1280x720 and framerate 15, Maybe the resolution is higher than optimal, because the cpu usage of the ffmpeg process is quite high on my pi 3 B+: mostly about 40%, but occasionally going as high as 80%. Do you think it is a normal cpu load for those parameters? Can you recommend any option for the ffmpeg command, which could reduce the cpu load?
Concerning the framerate, I didn't notice any difference in the Youtube livefeed, whether I set the parameter 'Webcam Frame Rate' to 5 or 15.

Best regards,
Tibor

ksztibor, 2019-02-14:
I've done a test with framerate 1. Then the cpu usage is only about 16%, so it is not the bottleneck for sure. However, in youtube the latency is still about 30 seconds, and still there are a lot of dropped frames: the picture is updated only about 3-4 times per minute (sometimes after 1 seconds, but mostly after about 20 seconds), instead of every second. What can be the problem?

ksztibor, 2019-02-14:
I've noticed, that the actual framerate of the mjpeg stream has to fit to the framerate settings in the ffmpeg command line. The problem is, that my webcam connected to the pi is set to 30 fps, and an ffmpeg framerate setting of 30 fps cannot be managed by the cpu. I've tried 15 fps, which is close to be manageable. This way the youtube stream is continuous. Any other ffmpeg framerate setting (e.g. 5 fps) results in choppy video again. I will keep on investigating on and experimenting with synchronizing the framerate of the mjpeg stream and the ffmpeg framerate settings, and various further ffmpeg options (e.g. -s, -maxrate, -minrate, -bufsize, -q, -preset, etc.) in order to find a good quality youtube streaming managable by the pi cpu.

By the way, the latency I still couldn't handle in any way.

adilinden, 2019-02-15:
Hi Tibor,

Because ffmpeg is performing transcoding, increasing the framerate will increase CPU usage. I've played with a few different ffmpeg options, but I haven't found anything that would reduce framerate further. I've tried to send a lower resolution stream to YouTube, but that caused streaming to fail altogether.

BTW, CPU usage for the streaming activity is around 40% on my rPi 2B.

I just recently discovered that the rPi cam supports higher resolution and faster framerates. I've toyed with the idea of running ffmpeg on another docker host dedicated to streaming to see what the results are. 

Adi

adilinden, 2019-02-15:
Hi Tibor,

I tried 1920x1080 with 10 fps. My rPi CPU spikes pretty bad with ffmpeg (inside) docker hogging the CPU. I am also getting errors in my YouTube Stream Health with frequent buffering.

> 8:34 PM Bad Video output low
YouTube is not receiving enough video to maintain smooth streaming. As such, viewers will experience buffering.
The stream's current bitrate (204.00 Kbps) is lower than the recommended bitrate. We recommend that you use a stream bitrate of 4500 Kbps.

I am not sure what YouTube standard processing delay is. I am getting about 30 seconds of delay between my actual video and the live stream. I have some other IP enabled cameras, using one of those and ffmpeg from command line (on a Mac, not rPi) results in a similar delay.

Just thought I'd share.

Adi

ksztibor, 2019-02-18:
Hi Adi,

thanks for your feedbacks. Not even after further investigation and experimenting I could find settings resulting in continuous youtube stream. Probably the mjpeg stream of my webcam (SQ11) is not fully suitable for h264 conversion. I've come to the conclusion, that for a good quality youtube streaming one needs a webcam with onboard h264 video compression, e.g. Logitech C920, so that the Pi only has to forward the video stream.

Best regards,
Tibor

closed 2019-02-18 by ksztibor

adilinden, 2019-02-18:
Hi @ksztibor 

OctoPi is my first ever exposure to RaspberryPi camera. Until working through the issues reported with the plugin, I wasn't even aware rPi could do higher resolution than 640x480. Having played with it now, I realize that using `ffmpeg` on rPi is definitely a compromise. I am personally using this plugin for now, but I am thinking of offloading the transcoding and live streaming from the rPi. 

I've done a brief test with the rPi cam at 1080p and 15fps. It seems to be just fine (without using the plugin and just viewing the mjpeg stream). 

My current thought is to develop a docker container that can on a datacenter style docker host to access the webcam stream, transcode, maybe even apply copyright free audio and some meaningful on screen display information (file name and or progress). This could then run 24/7, printing or not printing. Or it could be polling the OctoPrint API periodically to establish when the printer is printing and start and stop the stream automatically. 

Such a container could take anything supported by `ffmpeg` as an input stream, weather OctoPi cam, Ubiquity Unifi cam (which I happen to use and streamed just not for 3D printing), etc.

Adi