adilinden-oss/octoprint-webcamstreamer #7: Restart stream failed due to connectivity issues

adilinden, 2019-02-16:
When ffmpeg is unable to send data to ingest link it stops and container exists. Manual restart resumes stream.

adilinden, 2019-02-16:
Implemented and tested on container level using the [adilinden-oss/docker-rpi-stream](https://github.com/adilinden-oss/docker-rpi-stream) container. Added this to `entry.sh`:

```
    ffmpeg-loop)
        # Keep running ffmpeg until explicitely stopped
        shift
        while "true"; do
            echo "Executing \"ffmpeg $(echo $@ | cut -c -55)...\""
            ffmpeg $@

            # Pause for a bit before retrying
            echo -n "Pausing... "
            for ((wait=20; wait>=1; wait--))
            do
                echo -n "$wait "
                sleep 1
            done
            echo
        done
        ;;
```

To use, install the [adilinden/rpi-stream](https://cloud.docker.com/u/adilinden/repository/docker/adilinden/rpi-stream) on OctoPi command line (ssh session).

```
docker pull adilinden/rpi-stream:latest
```

Then change the WebcamStreamer Plugin settings to

- **Docker Image**: `adilinden/rpi-stream:latest`
- **FFmpeg Command**: `ffmpeg-loop -re -f mjpeg -framerate 5 -i {webcam_url} -ar 44100 -ac 2 -acodec pcm_s16le -f s16le -ac 2 -i /dev/zero -acodec aac -ab 128k -vcodec h264 -pix_fmt yuv420p -framerate {frame_rate} -g {gop_size} -strict experimental -filter:v {filter} -f flv {stream_url}`

TheSin-, 2019-10-16:
why hasn't this made it into a release yet, this is great I dislike haven't to reset mine everything the microwave is used ;)