adilinden-oss/octoprint-webcamstreamer #9: Low Resolution on Youtube Live

brooksben11, 2019-02-19:
Forgive me if this is a really stupid question (I'm not a software guy at all), but are there any suggestions for how I could increase the resolution on the Youtube Live stream I'm producing? I'm using a 3B+ with a rPi camera and while the local stream is of a reasonably high-resolution, the Youtube Live stream I'm producing is quite low. My actual internet connection isn't the issue as it's gigabit and I've also tested it out with the rPi sitting a few feet away from my router to eliminate any wifi issues. I've left everything at the defaults and also tried lowering the fps with no changes in the resolution. Thanks a ton in advance!

adilinden, 2019-02-22:
Hi @brooksben11 

Unfortunately the rPi is quite limited in its video processing abilities when it comes to transcoding the stream from mjpeg to flv. I've tried different resolutions and frame rates, but on my rPi 2B 640x480 at 5 fps is about the most it can handle. At those specs the YouTube or Twitch stream is still of noticeably worse quality than the native mjpeg stream. I also found that performance is best when rPi Cam settings are matching the live stream at 640x480 @ 5fps. While I don't have a rPi 3 to try, I don't think it performs that much better as H264 processing is already being offloaded to hardware. I am no expert on rPi hardware nor video processing, so these are really just my observation trying different settings.

Adi

brooksben11, 2019-02-23:
Thanks for the quick and detailed reply! Are there webcams that do all of this processing on-board that could be used instead of the rPi camera? Obviously I'm pretty naive when it comes to all of this... :) I'd just really love to have a high-quality stream as I really love the simplicity of your plugin!

adilinden, 2019-02-27:
Hi @brooksben11 

Unfortunately I don't have a lot of experience with rPi cameras and their best configuration. I had a  brief look last night as to how OctoPi handles cameras. I could be wrong, but it doesn't look like OctoPi supports anything other than USB cameras or rPi cameras. Doing a Google search reveals some feature requests but no solution. This means a IP based camera providing RTSP isn't supported in OctoPi and won't show an image in the Control tab.

Adi