adilinden-oss/octoprint-webcamstreamer #10: External docker server

K34nu, 2019-02-22:
Hi there,

Is it possible to specify a different docker server rather than hosted directly on the same container as OctoPrint? I've currently got my copy of octoprint running as a docker instance already, so it would be handy to run this as a seperate instance all together. 

Thanks,

adilinden, 2019-02-22:
Hi @K34nu 

There are several thoughts here. The plugin relies on the existence of a container that runs ffmpeg.

If OctoPrint is running as a container, then the plugin installed inside the same container could run the ffmpeg container on the same host, if the OctoPrint container has access to the hosts docker socket and docker environment variables are set properly.

Since the ffmpeg container accesses the MJPEG stream via its public URL, there is no reason that the ffmpeg container could not reside on a different docker host. This is in fact how I do development work on the underlying adilinden/rpi-ffmeg and adilinden/rpi-stream containers. I execute them manually on a different docker host and feeding them the webcam URL from my "production" 3D printer and OctoPi. That way I can develop with a single cam without impacting any printing activity that might be going on. This plugin right now does not support specifying a different host, although the python docker library does have the ability to specify a docker host. There is no way to handle authentication via the plugin that I could find. So all setup of docker hosts, authentication, would have to be handled manually outside of the plugin.

Personally, I am looking at developing a standalone streaming solution that will be able to take the OctoPi MJPEG stream as an input (one of many possible inputs) and convert it into a stream for live streaming. I am hoping to leverage the OctoPrint API to detect printer status. This would allow for starting and stopping of the stream with printing activity by polling the OctoPi at appropriate intervals. This may even allow for overlay of text that updates printing status, progress, etc... I do not believe that a rPi will have the capability to support this as its processing capabilities are too limited. Instead I'll be hosting the container in a commercial strength datacenter (my home lab). OctoPi and external docker host reside on the same LAN in my case. But systems can be geographically separated if appropriate firewall or VPN are used. I have no idea yet if there will be an OctoPrint plugin (not sure if there is anything to do for a plugin).

Those are my thoughts on the subject...

Adi

K34nu, 2019-02-22:
First off, thank you very much for the indepth reply! 

So just a few thoughts as I was reading it. 

> Since the ffmpeg container accesses the MJPEG stream via its public URL, there is no reason that the ffmpeg container could not reside on a different docker host.

So this is essentially what i'm looking for. I'm currently running the docker server on a mini pc hooked up to the printer via usb, but my whole idea behind it all was to run multiple octoprint containers which all control individual printers (Currently have 2, looking at a third). Would there be any reason I couldn't use one of the containers you specified? (adilinden/rpi-ffmeg and adilinden/rpi-stream)

> Personally, I am looking at developing a standalone streaming solution that will be able to take the OctoPi MJPEG stream as an input

That sounds brilliant. I've been battling for months trying to figure out a way to get the rpi ones that you based your image off working (Running docker on x86). So i'd be really interested in this. I'm pretty new to github though, is there a way that I can subscribe to your work so I can stay up to date on your progress with this please? 

Thank you again for your indepth reply, I really appreciate you taking your time out to write it. Wish you the best of luck with your project!

adilinden, 2019-02-22:
Hi @K34nu 

The adilinden/rpi-ffmeg and adilinden/rpi-stream containers were built on rPI and are ARM based. They won't run on x86. I intended for adilinden/rpi-ffmeg to be a dockerized ffmpeg. Then I am using the adilinden/rpi-stream to refer to adilinden/rpi-ffmeg and wrap it into shell script or whatever else I think might work. It saves a lot of time not having to rebuilt ffmpeg with every minor change.

I already created adilinden/ffmeg, or adilinden-oss/docker-ffmpeg here on GitHub. I briefly tested it and it seems to work. Next will be the container that does all of the logic. I have a good idea what I want it to accomplish but no firm plan yet on how to do it (like shell scripting or python, or?).

I am not sure how one can follow a repo on GitHub, I never tried.

Adi

adilinden, 2019-03-04:
Here is what I am working on. Still pretty rough right now.
[adilinden-oss/docker-octoprint-streamer](https://github.com/adilinden-oss/docker-octoprint-streamer)

K34nu, 2019-03-04:
Oh awesome! Thank you! I'll give that a go and let you know how I get along with it! I've currently taken to dropping the feed directly into OBS via the "Browser" part. Not ideal, but it works. Do I need to do anything different / make any changes to the octoprint plugin for this to work?

Thanks,

adilinden, 2019-03-04:
It doesn't use the OctoPrint plugin at all. It is not compiled for ARM either, but requires docker on a x86 (Intel) host. There is no documentation yet either. Will have to work on that later this week. I've had a lass polished setup, as in API keys and URL coded inline and no on-screen text, working for well over a week. This is really taking that work, adding some features (more background music and on-screen printer and job data) all the while getting it into a format that can be shared.

So it is coming along, but definitely needs a bit more work.