adilinden-oss/octoprint-webcamstreamer #11: Facebook Streaming

Hotwire, 2019-02-26:
I was able to get facebook streaming to work last night.. sort of. Has anyone else been trying to get this to work? I will try to recreate my steps tonight, so that I can document it here.

victor-ourd, 2020-01-22:
i've try, dont work...

JustinBenedick, 2020-04-05:
im trying to get facebook streaming going but its not working. how to do I put the url and stream key together?

apbarratt, 2020-07-24:
Anyone else had any success with this?
I’ve tried the stream url and key together for the URL and just took a guess for the embed URL (don’t really need it anyway as I’d just watch on Facebook Producer).
It happily says that it has started streaming, but Facebook Producer just sits saying it’s waiting for a live stream.