adilinden-oss/octoprint-webcamstreamer #12: Add Non-Pi Support

jneilliii, 2019-02-27:
I was able to get the plugin to work with a different docker image for ffmpeg on windows, would probably work for linux distributions as well.  You can find my docker image over [here](https://github.com/jneilliii/docker-ubuntu-stream).  I also signed up on docker and it can be pulled via command `docker pull jneilliii/ubuntu-stream`.  Not sure if there is a way to conditionally load one reference over another in docker, but that's all I ended up having to change to make this x86 compatible versus the ARM compile of ffmpeg. For windows the only other thing I had to do was manually `pip install pypiwin32` to resolve some npipe errors I was getting with the plugin.  Not sure if you could conditionally include that as well based on OS in setup.py.

adilinden, 2019-03-04:
I do have ffmpeg on x86 as well. Just managed to make it even smaller by building it on Alpine.
Github [adilinden-oss/docker-ffmpeg](https://github.com/adilinden-oss/docker-ffmpeg) and Docker Hub [adilinden/ffmpeg](https://hub.docker.com/r/adilinden/ffmpeg)

adilinden, 2019-03-04:
I am also working on using OctoPi for physical connection to the printer, but streaming off-loaded to a docker host in a "datacenter" (realistically my homelab). That's beginning to take shape here [adilinden-oss/docker-octoprint-streamer](https://github.com/adilinden-oss/docker-octoprint-streamer). I don't think that there is going to be any integration as a plugin with OctoPrint in the near future. It just access the MJPEG Stream and sends it away to YouTube or Twitch.

I made some interesting observations, like realizing that the 640x480 MJPEG stream at 10 frames per seconds results in a 15Mbps datastream! Using this type of model one quickly hits the limit of USB wireless dongles. Bumbing things up to 720p isn't really an option.

I also learned that the `ffmpeg` process requires a lot of CPU horsepower. Processing the 640x480@10fps stream takes close to 100% of a single core/CPU on my docker host VM. I think that it's a 2.4GHz Xeon that powers it. Bumping things to 720p also causes a significant increase in CPU in addition to troubles with wireless throughput.

jneilliii, 2019-03-04:
That's interesting results. Did you try with h264_omx driver for the vcodec encoding in the ffmpeg command? That should offload to the GPU if i'm not mistaken.

adilinden, 2019-03-04:
The rpi-ffmpeg container does have h264_omx support. The non-rpi ffmpeg container is meant for x86. I am pretty sure h264_omx is rPi specific and not an available feature on x86.

jneilliii, 2019-03-04:
Yes, that's true.  I thought that's what you meant in regards to CPU utilization that it was on the pi since your default command line is using just h264.  I wonder if there are equivalent GPU vcodecs for x86 devices that have a GPU as well that could be utilized.  I just don't know enough about ffmpeg to know.