adilinden-oss/octoprint-webcamstreamer #14: Possibility for music or other audio?

Napa39, 2019-03-01:
All the streams from this plugin are dead silent, and that's not usually fun for some viewers. Love what it does (wish my webcam could output more than 5 frames a second, but that'll be a future purchase), but the lack of audio is what's lacking from the plugin in my opinion. Would it be possible to have it stream music (from Monstercat, for example) as a setting?

clumsyzombie, 2019-07-11:
Man this would be awesome if you could do some kind of music or something that would be super cool. I have not figured out how to do that with out having a true computer attached.

Vertabreak, 2019-10-26:
my camera has a built in mic is there any way to make use of it for commentary while streaming?