adilinden-oss/octoprint-webcamstreamer #22: Feature request, overlay or/and second stream

TheSin-, 2019-09-11:
I'd love to spice up my twitch, be amazing to overlay some of the print info into the stream, temps, eta, etc etc. I could make a mock up.  Be nice to have a second stream too so I could put a webcam in the corner like gamers do so I can talk with others while it streams.

Of course these are just ideas wanted to jot them down and see if it's possible.

TheSin-, 2019-10-16:
so I was thinking of something like

```
self.overlay_cmd_default = "-i /tmp/stream_overlay.png -filter_complex '[0:v][1:v] overlay=0:0'"
```

the png path would need to be inside the locker image, so using /var/lib/docker to write it out on the local system, or add a mount point in the docker init call so we can write out the image every x seconds.

the overlay=0:0 is where to place the overlay on the stream x:y based on top left.  This might need to be a variable or static to 0:0 and just make it so the image generator take in account the size of the video.

I started trying to mod things but I'm not versed well in docker or the octoprint plugin framework, so figured I'd post things here cause maybe someone could make this work faster then me.  I'd love to be able to use things like nozzle temp, bed temp, elapsed time, time remaining, maybe some icons and such, maybe include a few icons, to use in front of items.

Could allow upload of images that could be placed on a mat, then just have the image generate on x interval, which would just be accessible to the docker image and get added to the stream, but default it could just be a blank png so nothing changes by default?

TheSin-, 2020-03-30:
Not from me I haven't hate time to mess with it, but I plan to look into it soon, Hopefully if I put in a PR it'll get more response from the maintainer.

TheSin-, 2021-12-30:
here is my progress so far

https://github.com/TheSin-/octoprint-webcamstreamer/tree/master

adilinden, 2022-02-11:
I have done this at https://github.com/adilinden-oss/docker-octoprint-streamer, but don't mistake the name as it **will not run on rPi**. I might see if this could be simplified and work on rPi.

closed 2022-02-11 by adilinden