adilinden-oss/octoprint-webcamstreamer #23: Docker container not starting

muldaddy, 2019-09-28:
When I hit "Go Live" in the WebCamStreamer plugin page in Octoprint, it would give me positive feedback, but my Twitch stream was offline.  

Running
  `docker run --name WebStreamer -ti adilinden/rpi-ffmpeg ffmpeg -re -f mjpeg -framerate 5 -i http://192.168.1.137/webcam/?action=stream  -ar 44100 -ac 2 -acodec pcm_s16le -f s16le -ac 2 -i /dev/zero -acodec aac -ab 128k -vcodec h264 -pix_fmt yuv420p -framerate 5 -g 10 -strict experimental -f flv rtmp://live-den.twitch.tv/app/live_XXXXX`

from the CLI gave me an error '* failed to open vchiq instance'.

Google'ing suggested adding a `--device=/dev/vchiq` option to the docker run command and lo and behold! 

I know just enough Python to be dangerous, but I felt brave so I edited the ~/OctoPrint/venv/lib/python2.7/site-packages/octoprint_webcamstreamer/__init__.py file.

At Line 190:
`privileged = False,
devices=["/dev/vchiq"]`

I restarted the OctoPrint server (necessary?) and now I when I hit "Go Live!" it really does.

I don't know Git well enough to add a commit, but I wanted to comment so others might find this.

iandouglas, 2019-10-09:
Nice find. This didn't work for me on a Pi 4. If I manually run the docker command like yours above (adjusting the IP and streaming URL, etc), then it streams, but I have to shell into the Pi for that to work.

curiosus0ne, 2020-01-04:
Hi, i tried to fix the issue I had (same as yours) by modifying the privileged setting to no avail unfortunately... [image]

adilinden, 2022-02-11:
https://github.com/adilinden-oss/docker-rpi-ffmpeg shows how to start the stream manual via CLI (ssh to OctoPrint server). 

To use simply execute `ffmpeg`.

```
docker run -it --rm --privileged adilinden/rpi-ffmpeg ffmpeg \
    -re -f mjpeg -framerate 5 -i http://<IP of OctoPi>:8080/?action=stream \
    -ar 44100 -ac 2 -acodec pcm_s16le -f s16le -ac 2 -i /dev/zero -acodec aac -ab 128k \
    -strict experimental -vcodec h264 -pix_fmt yuv420p -g 10 -vb 700k -framerate 5 \
    -f flv rtmp://a.rtmp.youtube.com/live2/<YouTube Stream ID>
```

closed 2022-02-11 by adilinden