adilinden-oss/octoprint-webcamstreamer #30: 'NoneType' object has no attribute 'containers'

sonatamartica, 2020-04-09:
I've entered all the right info but when I open octoprint I get an "unknown protocol" error with "livehttps" and when I click go live it pops up "WebcamStreamer Error
'NoneType' object has no attribute 'containers'".

azzuron, 2020-04-29:
I get this error also and i do not have the FFMPEG container installed. I already run octoprint in a container on a server. My container includes ffmpeg in it and i think its still looking for the container for ffpmeg to use to do the encoding for the stream. I want to use the ffmpeg included with my container but im not sure how i can do that with the options given.

MacCracken, 2020-05-28:
@sonatamartica You have to manually install docker and make sure the pi user has access.  Then reboot Octoprint; as it worked for me after reboot.

adilinden, 2022-02-11:
Sorry, this plugin is being developed with OctoPi and OctoPrint running on bare metal. I have not run OctoPrint in a container and no real need to. Currently the plugin requires that `ffmpeg` is run as a dedicated container as the plugin looks for and starts and stops the container, not a `ffmpeg` process.

closed 2022-02-11 by adilinden