adilinden-oss/octoprint-webcamstreamer #32: Whoops! This embed is misconfigured.  (Developers: Please check your browser console for more information)

coolguymatt, 2020-07-16:
Hi,

I believe the way the embed link is processed is incorrect now. It looks like Twitch changed how it works on June 10 (https://discuss.dev.twitch.tv/t/twitch-embedded-player-updates-in-2020/23956). You need to specify the parent now.

If I check the Developer console in Chrome I see this error:
`[NoParent] parent query string value was not specified`

coolguymatt, 2020-07-16:
I figured this out.

- you need to use an https URL
- the embed link is now this format: https://player.twitch.tv/?channel=CHANNEL&parent=OCTOPRINTURL (if you have multiple URLs you access octoprint from, you can add the following - &parent=OCTOPRINTURLALT )

closed 2020-07-16 by coolguymatt

derekpurdy, 2020-08-17:
@coolguymatt, I followed your example but I now get "player.twitch.tv refused to connect."