adilinden-oss/octoprint-webcamstreamer #33: YouTube "Please use a keyframe frequency of four seconds or less" error from one printer but not another

KenLucke, 2020-08-30:
When streaming from one printer, everything works perfectly.  When streaming from an identical printer, set up identically, with all the same settings, I consistently get "Poor connection" complaints from YouTube, with the following error message banded across my streaming setup page:

`Please use a keyframe frequency of four seconds or less. Currently, keyframes are not being sent often enough, which will cause buffering. The current keyframe frequency is 14.6 seconds. Note that ingestion errors can cause incorrect GOP (group of pictures) sizes.`

I've tried changing the frame rate in the plugin, but that does not eliminate the error, nor does "frame rate" seem to be the same as "keyframe rate" in the error.

Where do I fix this?

Also, on EITHER printer, when clicking on the red "Play" button in the plugin tab, I get the following type of error every time:

`An error occurred. Please try again later. (Playback ID: SkfGB06f3bruk6Pv)
Learn More`

Tihis entire text is a hyperlink,  Clicking on the text link, I am taken to a generic YouTube "Troubleshoot Video Streaming Issues" page, which, as with most YT help pages, is pathetically unhelpful.  There is also a gray "play" type arrow appear at the bottom right of the video frame, which just takes me to the YouTube home page.

mjrider, 2020-09-22:
The framerate/gop problems could be related to this https://github.com/adilinden-oss/octoprint-webcamstreamer/pull/26

adilinden, 2022-02-11:
I believe this should be fixed now.

closed 2022-02-11 by adilinden