adilinden-oss/octoprint-webcamstreamer #36: Error when press "Go Live"

Nuttavoot, 2020-10-14:
This is the error I got "
WebcamStreamer Error
404 Client Error: Not Found ("pull access denied for adilinden/rpi-ffmpeg, repository does not exist or may require 'docker login': denied: requested access to the resource is denied")"

But successfully run "docker run hello-world"

ppetrov-dev, 2020-10-29:
I have the same error. Let's wait for the author together=)

JoshJamesLS, 2020-10-30:
After having the same issue time and time again, I done some digging. In this, I found that the "rpi-ffmpeg" docker file no longer exists on adilinden's docker page. If this is intentional by adilinden or not. But it certainly prevents anyone from setting up the webcam streamer in the future. With any luck, this is a mistake, he will see this, and then he will re-upload it.

kolisko, 2020-11-09:
I fixed the rpi-ffmpeg Dockerfile and created the pull request https://github.com/adilinden-oss/docker-ffmpeg/pull/1. Until PR will not be merged here is the workaround - you can pull docker image kolisko/rpi-ffmpeg instead the missing official one and change the WebcamStreamer Plugin/Advanced Settings/Docker Image to that image kolisko/rpi-ffmpeg.

ttrueten, 2020-12-16:
docker pull kolisko/rpi-ffmpeg runs well. Thank you.

T.

closed 2022-02-11 by adilinden