adilinden-oss/octoprint-webcamstreamer #37: Unable to pull Docker image

EvilBlood, 2020-10-31:
If i try to pull the Docker-image and i get the following error: 

Error response from daemon: pull access denied for adilinden/rpi-ffmpeg, repository does not exist or may require 'docker login': denied: requested access to the resource is denied

LukeLevesque, 2020-11-02:
I'm having the exact same issue following the setup guide.

JoshJamesLS, 2020-11-06:
https://github.com/adilinden-oss/octoprint-webcamstreamer/issues/36#issuecomment-719594187

JoshJamesLS, 2020-11-06:
@adilinden

kolisko, 2020-11-09:
I fixed the rpi-ffmpeg Dockerfile and created the pull request adilinden-oss/docker-ffmpeg#1. Until PR will not be merged here is the workaround - you can pull docker image kolisko/rpi-ffmpeg instead the missing official one and change the WebcamStreamer Plugin/Advanced Settings/Docker Image to that image kolisko/rpi-ffmpeg.

orichienal, 2020-12-06:
hi, 
i there another docker source for the "adilinden/rpi-stream"?
thx

EnglishAW, 2020-12-18:
@kolisko Hey I am trying to get this to work on my RPi 2B. I was able to pull your docker image and adjust the setting on the plugin to use that image. When I hit go live it still does not start streaming. I was wondering if it was having trouble starting the image so I tried to do a docker run on your image manually and got this error `WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm/v7) and no specific platform was requested
standard_init_linux.go:219: exec user process caused: exec format error`

Do you know what might be going wrong here. Thank you for any help that you can offer.

kolisko, 2020-12-18:
@EnglishAW Hi, yes I know what is wrong. :) I built the image only for AMD CPUs. So I newly rebuilt the image for multiple CPU platforms - linux/amd64, linux/arm64, linux/386, linux/arm/v7, linux/arm/v6. 

So you should update the image:

`docker pull kolisko/rpi-ffmpeg:latest`

here you can check if you have right digest:

https://hub.docker.com/r/kolisko/rpi-ffmpeg/tags?page=1&ordering=last_updated

kolisko, 2020-12-18:
@orichienal Hi, I also prepared kolisko/rpi-stream image for you

EnglishAW, 2020-12-19:
@kolisko OMG! Thank you for the fast response and actually doing this for me. I love this community. With the new docker image I am able to run it and see that some data is being passed to twitch and youtube but both report unstable stream and fail to maintain the stream. 

I am wondering if you know of any configuration either with ffmpeg or octoprint that could give a more stable bitrate. I have tried to increase the framerate on both ffmpeg and in the /boot/octopi.txt but have had no success. I am worried that perhaps the RPi 2B just cannot keep up with the demand of streaming. If I can deduce that this is indeed the problem I have no problem ordering a newer RPi. 

Thanks again for any guidance you have to offer!

kolisko, 2020-12-19:
@EnglishAW The problem could be the performance of RPi 2B and your internet upstream bandwidth. Octoprint recommends:

_Recommended hardware: Raspberry Pi 3B, 3B+ or 4B. Expect print artifacts and long loading times with other options, especially when adding a webcam or installing third party plugins_

 I am using RPi 4.

arnehude, 2020-12-20:
Thank you @kolisko, i just got it working on RPi4

adilinden, 2022-02-11:
https://hub.docker.com/r/adilinden/rpi-ffmpeg exists again.

adilinden, 2022-02-11:
> @EnglishAW The problem could be the performance of RPi 2B and your internet upstream bandwidth. Octoprint recommends:
>
> _Recommended hardware: Raspberry Pi 3B, 3B+ or 4B. Expect print artifacts and [...]

I am using a Raspberry Pi 2 Model B Rev 1.1

closed 2022-02-11 by adilinden