adilinden-oss/octoprint-webcamstreamer #39: repository does not exist or may require 'docker login'

ttrueten, 2020-12-09:
Hello,
i get the following response: 
`pi@octopi:~ $ docker pull adilinden/rpi-stream:latest
Error response from daemon: pull access denied for adilinden/rpi-stream, repository does not exist or may require 'docker login': denied: requested access to the resource is denied`

Any hints?

Cheers,
Thomas

spence572, 2020-12-15:
https://github.com/adilinden-oss/octoprint-webcamstreamer/issues/36#issuecomment-723980053

kolisko, 2020-12-18:
@ttrueten I created the PR adilinden-oss/docker-ffmpeg#1 fix and built the kolisko/rpi-stream image

CobaltEcho, 2021-01-02:
@kolisko Is there a way to install your fix and use it currently?

kolisko, 2021-01-02:
@CobaltEcho Yes, it should work exactly as you described

foosel, 2021-02-18:
We also just got a report about this on the OctoPrint plugin repository:

https://github.com/OctoPrint/plugins.octoprint.org/issues/784

@adilinden-oss are you still actively maintaining this plugin? If so, could you please either fix your docker repository or update the README and your plugin listing to a working state? Thank you.

adilinden, 2022-02-11:
https://hub.docker.com/r/adilinden/rpi-ffmpeg exists again and has working images.

closed 2022-02-11 by adilinden