adilinden-oss/octoprint-webcamstreamer #40: Stream without docker?

briandilley, 2021-01-20:
Is there a way to stream without using docker?  If not, i'd like to make that as a feature request.

Octopi comes with ffmpeg installed already, and it's compiled with hardware acceleration support for the raspberry pi. I'd like to just use that ffmpeg directly rather than a docker container if possible.

What was the reason for doing it in a Docker container anyway? (No shade, genuinely curious)

orchestrateddigital, 2021-01-24:
Was going to ask the same. Seems pretty try-hardy for no reason.

adilinden, 2022-02-10:
I would like to accomplish this, but when I wrote the plugin I couldn't figure out how to start and stop the ffmpeg process running on the host directly. Identifying the docker container was relatively easy.